IcyFeather233/ospp23-rtt #14: Task: 尝试给一些bsp添加Kconfig .config等文件，方便menuconfig

IcyFeather233, 2023-05-23:
issue 地址：

https://github.com/RT-Thread/rt-thread/issues/7530

IcyFeather233, 2023-05-23:
# 应用：在项目中使用 env

## 使用 env 的要求

- menuconfig 是 RT-Thread 3.0 以上版本的特性，推荐将 RT-Thread 更新到 3.0 以上版本。
- 目前 RT-Thread 还没有对所有的 bsp 做 menuconfig 的支持，也就是说有些 bsp 暂时还不能使用 menuconfig 来进行配置，但常用的 bsp 都已经支持。

## menuconfig 中选项的修改方法

如果想在 menuconfig 的配置项中添加宏定义，则可以修改 bsp 下的 Kconfig 文件，修改方法可以在网络中搜索Kconfig语法关键字获得详细的说明文档，也可以参考 RT-Thread 中的 Kconfig 文件或者已经支持过 menuconfig 的 bsp 中的 Kconfig 文件。
新的项目添加 menuconfig 功能

这里的新项目指的是，还未生成 .config 和 rtconfig.h 的全新开发的项目。因为这两个文件，只有在 menuconfig 第一次保存时才会创建。具体流程如下：

- 将已经支持 menuconfig 功能的 bsp 里面的 kconfig 文件拷贝到新的项目根目录中。
- 注意修改 Kconfig 中的 RTT_ROOT 值为 RT-Thread 所在目录，否则可能提示找不到 RTT_ROOT 。
- 使用 menuconfig 命令开始配置即可。

## 旧项目添加 menuconfig 功能

这里的旧项目指的是已经经过一段时间的开发，而且项目中存在已经修改过的 rtconfig.h文件 ，但是没有使用过 menuconfig 来配置的项目。具体流程如下：

- 首先备份旧项目内的 rtconfig.h 文件。
- 使用 scons —genconfig 命令根据已有的 rtconfig.h 生成 .config 文件，这里生成的 .config 文件保存了旧项目中 rtconfig.h 文件对项目的配置参数。
- 将已经支持 menuconfig 功能的 bsp 里面的 kconfig 文件拷贝到要修改项目的根目录中。
- 使用 menuconfig 命令来配置我们要修改的旧项目。menuconfig 会读取第二步生成的 .config 文件，并根据旧项目的配置参数生成新的 .config 文件和 rtconfig.h 文件 。
- 对比检查新旧两份 rtconfig.h 文件，如果有不一致的地方，可以使用 menuconfig 命令对配置项进行调整。

参考资料： https://www.bookstack.cn/read/RT-Thread-tool/1.md

IcyFeather233, 2023-05-23:
先测试一下看看：

```
IcyFeather@DESKTOP-C2V9VAL C:\Users\IcyFeather\Desktop\rt-thread\bsp\lpc178x
> scons                                                                     
scons: Reading SConscript files ...                                         
Newlib version: 4.1.0                                                       
scons: done reading SConscript files.                                       
scons: Building targets ...                                                 
scons: building associated VariantDir targets: build                        
CC build\applications\application.o                                         
CC build\applications\startup.o                                             
CC build\CMSIS\CM3\CoreSupport\core_cm3.o                                   
AS build\CMSIS\CM3\DeviceSupport\NXP\LPC177x_8x\startup\gcc\startup_LPC177x_
CC build\CMSIS\CM3\DeviceSupport\NXP\LPC177x_8x\system_LPC177x_8x.o         
CC build\drivers\board.o                                                    
CC build\drivers\drv_glcd.o                                                 
CC build\drivers\lpc177x_8x_clkpwr.o                                        
CC build\drivers\lpc177x_8x_emc.o                                           
CC build\drivers\lpc177x_8x_pinsel.o                                        
CC build\drivers\lpc177x_8x_uart.o                                          
CC build\drivers\lpc17xx_lcd.o                                              
CC build\drivers\sdram.o                                                    
CC build\drivers\uart.o                                                     
CC build\kernel\components\finsh\msh.o                                      
CC build\kernel\components\finsh\msh_parse.o                                
CC build\kernel\components\finsh\shell.o                                    
CC build\kernel\components\libc\compilers\common\cctype.o                   
CC build\kernel\components\libc\compilers\common\cstdio.o                   
CC build\kernel\components\libc\compilers\common\cstdlib.o                  
CC build\kernel\components\libc\compilers\common\cstring.o                  
CC build\kernel\components\libc\compilers\common\ctime.o                    
CC build\kernel\components\libc\compilers\common\cwchar.o                   
CC build\kernel\components\libc\compilers\newlib\syscalls.o                 
CC build\kernel\libcpu\arm\common\div0.o                                    
CC build\kernel\libcpu\arm\common\showmem.o                                 
AS build\kernel\libcpu\arm\cortex-m3\context_gcc.o                          
CC build\kernel\libcpu\arm\cortex-m3\cpuport.o                              
CC build\kernel\src\clock.o                                                 
CC build\kernel\src\components.o                                            
CC build\kernel\src\device.o                                                
CC build\kernel\src\idle.o                                                  
CC build\kernel\src\ipc.o                                                   
CC build\kernel\src\irq.o                                                   
CC build\kernel\src\kservice.o                                              
CC build\kernel\src\mem.o                                                   
CC build\kernel\src\memheap.o                                               
CC build\kernel\src\mempool.o                                               
CC build\kernel\src\object.o                                                
CC build\kernel\src\scheduler_up.o                                          
CC build\kernel\src\thread.o                                                
CC build\kernel\src\timer.o                                                 
LINK rtthread-lpc178x.elf                                                   
arm-none-eabi-objcopy -O binary rtthread-lpc178x.elf rtthread.bin           
arm-none-eabi-size rtthread-lpc178x.elf                                     
   text    data     bss     dec     hex filename                            
  49376    1288    2344   53008    cf10 rtthread-lpc178x.elf                
scons: done building targets.                                               
                                                                            
IcyFeather@DESKTOP-C2V9VAL C:\Users\IcyFeather\Desktop\rt-thread\bsp\lpc178x
> menuconfig                                                                
                                                                            
<menuconfig> 命令应当在某一特定 BSP 目录下执行，例如："rt-thread/bsp/stm32/stm32f091-st-nucleo"
请确保当前目录为 BSP 根目录，并且该目录中有 Kconfig 文件。                                        
                                                                            
<menuconfig> command should be used in a bsp root path with a Kconfig file. 
Example: "rt-thread/bsp/stm32/stm32f091-st-nucleo"                          
You should check if there is a Kconfig file in your bsp root first.         
```

发现 scons 编译正常，但是 menuconfig 无法运行，提示需要 Kconfig 文件

supperthomas, 2023-05-23:
可以参考其他bsp添加简单的Kconfig

IcyFeather233, 2023-05-23:
目前看来最简单的方法为，在缺失 Kconfig 文件的 bsp 目录下添加 Kconfig，里面填这些内容：

```
mainmenu "RT-Thread Project Configuration"

config BSP_DIR
    string
    option env="BSP_ROOT"
    default "."

# 这里面的 default 需要根据实际情况改变，如果是 rt-thread\bsp\lm3s9b9x 这样的层级就是两级，如果是三级就写成 default "../../.."
config RTT_DIR
    string
    option env="RTT_ROOT"
    default "../.."

config PKGS_DIR
    string
    option env="PKGS_ROOT"
    default "packages"

source "$RTT_DIR/Kconfig"
source "$PKGS_DIR/Kconfig"
```

但是实际上其余 bsp 的 Kconfig 基本都是类似这样：

```
mainmenu "RT-Thread Project Configuration"

config BSP_DIR
    string
    option env="BSP_ROOT"
    default "."

config RTT_DIR
    string
    option env="RTT_ROOT"
    default "../.."

config PKGS_DIR
    string
    option env="PKGS_ROOT"
    default "packages"

source "$RTT_DIR/Kconfig"
source "$PKGS_DIR/Kconfig"

config SOC_LPC4088
    bool
    select ARCH_ARM_CORTEX_M4
    select RT_USING_COMPONENTS_INIT
    select RT_USING_USER_MAIN
    default y

source "$BSP_DIR/drivers/Kconfig"
```

也即加了一个：

```
# SOC_型号
config SOC_LPC4088
    bool
# ARCH需要查相应型号资料对应填写
    select ARCH_ARM_CORTEX_M4
    select RT_USING_COMPONENTS_INIT
    select RT_USING_USER_MAIN
    default y
```

不知道这个部分是否必要，且目前没有搞清楚这个部分的具体作用。

supperthomas, 2023-05-23:
.config 的作用是生成rtconfig.h 只要你最后生成的rtconfig.h 能够覆盖之前的即可。这里可以找下相关教程看下。

IcyFeather233, 2023-05-23:
[Kconfig 相关语法整理](https://www.jianshu.com/p/aba588d380c2)

IcyFeather233, 2023-05-23:
新添加了 Kconfig 并运行 menuconfig 修改配置后，选择保存会出现：

[image]

退出后再次运行 menuconfig 发现配置已保存

猜测 Kconfig、.config、rtconfig.h 三个文件的作用：

- Kconfig
提供 menuconfig 运行后的图形化配置
- .config
.config 是通过 menuconfig 自动生成的，menuconfig 中未手动修改的宏就是 Kconfig 中的默认值，.config 把所有用到的宏的值都以类似 `CONFIG_RT_NAME_MAX=8` 的形式一一声明出来
- rtconfig.h
rtconfig.h 通过 .config 生成，把 .config 中 `CONFIG_RT_NAME_MAX=8` 的形式换成了 `#define RT_NAME_MAX 8` 的形式，便于 C 程序读取

IcyFeather233, 2023-05-23:
目前感觉这个 issue 的解决方式就是：
- 找到所有缺少 Kconfig 的 bsp 目录
- 手动补上 Kconfig （按照上面的模板）
- 运行 menuconfig 自动生成 .config 文件
- 测试一下 menuconfig 中修改配置项是否能成功修改（不知道这个是否需要测？会存在修改失败的情况吗？）

supperthomas, 2023-05-23:
> 目前感觉这个 issue 的解决方式就是：
> 
> * 找到所有缺少 Kconfig 的 bsp 目录
> * 手动补上 Kconfig （按照上面的模板）
> * 运行 menuconfig 自动生成 .config 文件
> * 测试一下 menuconfig 中修改配置项是否能成功修改（不知道这个是否需要测？会存在修改失败的情况吗？）

需要用scons编译一下，还有看下rtconfig.h上的宏是否都有覆盖。

supperthomas, 2023-05-23:
可以先提交一个试试看。

IcyFeather233, 2023-05-23:
https://github.com/RT-Thread/rt-thread/actions/runs/5014814877/jobs/8989555408 中缺少 .config 的有：

- lpc178x [fix]
- lpc2148 [fix]
- lpc2478 [fix]
- lpc5410x [fix]
- lpc176x [fix]
- xplorer4330/M4 [fix]

supperthomas, 2023-05-23:
看下改动前keil工程中的文件和改动后的keil工程，看下那些文件多出来了，看看这些多的文件因为哪个宏多出来的，尝试把这个宏去掉，先找到关键的宏

IcyFeather233, 2023-05-24:
> https://github.com/RT-Thread/rt-thread/actions/runs/5014814877/jobs/8989555408 中缺少 .config 的有：
> 
>     * lpc178x [fix]
> 
>     * lpc2148 [fix]
> 
>     * lpc2478 [fix]
> 
>     * lpc5410x [fix]
> 
>     * lpc176x [fix]
> 
>     * xplorer4330/M4 [fix]

针对这里面 Kconfig .config 写完后，menuconfig 可以正常使用但是编译失败的问题，我在这篇文章里进行了详细的说明：
https://club.rt-thread.org/ask/article/f28fc6edf3877f4b.html